HelTecAutomation/CubeCell-Arduino #124: .S files: no preprocessing

eiten, 2020-08-14:
code like `#ifdef __ASR6501__` are not processed in assembler files. Please change [this lines](https://github.com/HelTecAutomation/ASR650x-Arduino/blob/fb4af3373da61189ca049559fa9eccd7944f95d9/platform.txt#L29-L30) in platform txt to
```
compiler.S.cmd=arm-none-eabi-gcc
compiler.S.flags=-mcpu=cortex-m0plus -mthumb -c -x ass Iembler-with-cpp -g -w -gdwarf-2 "-I{compiler.sdk.path}/projects/PSoC4"
```
to fix the issue. It's the way it's done in other cortex based Arduino implementations

Heltec-Aaron-Lee, 2020-08-14:
Did you tested that?  I tried as yours,   error occurred.
[image]

eiten, 2020-08-14:
My bad! Sorry, that was a copy paste error! It should read 
```
compiler.S.cmd=arm-none-eabi-gcc
compiler.S.flags=-mcpu=cortex-m0plus -mthumb -c -x assembler-with-cpp -g -w -gdwarf-2 "-I{compiler.sdk.path}/projects/PSoC4"
```

Heltec-Aaron-Lee, 2020-08-14:
Indeed the 'compiler.S.cmd=arm-none-eabi-as' refer from the Cypress original example, the .S files are automatically generated by IDE. So we just keep this part the same as Cypress one.

We are not very professional in compilation principle, can you please tell me why?

eiten, 2020-08-14:
Of course, for example if you look at [this file](https://github.com/eiten/MySensors/blob/development/MyASM.S) (I am writing a CubeCell hal for the MySensors library).
if you use `arm-none-eabi-as` in stead of `arm-none-eabi-gcc`, it will crash on [line 32](https://github.com/eiten/MySensors/blob/08391cabc3b44025e9d4c83146bcba9470358dd7/MyASM.S#L31) because `-as` does not process the preprocessor directives in lines [21](https://github.com/eiten/MySensors/blob/08391cabc3b44025e9d4c83146bcba9470358dd7/MyASM.S#L21) and [28](https://github.com/eiten/MySensors/blob/08391cabc3b44025e9d4c83146bcba9470358dd7/MyASM.S#L28).

eiten, 2020-08-14:
Btw, it's the standard in cortex arduino frameworks, for example the [Adafruit_nRF52](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/3951aaab691613276b707acf69c463dc9181acc5/platform.txt#L36-L37) implementation, [SAM3X](https://github.com/arduino/ArduinoCore-sam/blob/eed66e7977b4037f49bbcd832437d4bcf72c0f3f/platform.txt#L25-L26)/[SAMd21](https://github.com/arduino/ArduinoCore-samd/blob/c5033b393509af0194534388f7c9cec2885fe0fa/platform.txt#L45-L46), [XMC1100](https://github.com/Infineon/XMC-for-Arduino/blob/9a4e6c534b6c33b7656c67483228f9c8179ebe47/platform.txt#L33) , [cc3200](https://github.com/energia/cc3200-core/blob/c5507401fe46454d38c3ce3af0a140df8fc088a3/platform.txt#L25-L26) or [STM32](https://github.com/stm32duino/Arduino_Core_STM32/blob/59dda5c138badcb0a17a24f7a30fa81dececa0d7/platform.txt#L22).
In fact, I never came across a Cortex where this is solved different.

Heltec-Aaron-Lee, 2020-08-19:
Thank you for your hint. I had fixed that.

closed 2020-08-19 by Heltec-Aaron-Lee