HelTecAutomation/CubeCell-Arduino #130: Missing AT_user_check

aparcar, 2020-09-04:
The [AT_Command](https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/libraries/LoRa/examples/AT_Command/AT_Command.ino) no longer contains the user specific `AT_user_check` method, this should be re-added for clarity.

Heltec-Aaron-Lee, 2020-09-18:
What's the function of `AT_user_check`? Can you please provide some more information?

aparcar, 2021-01-11:
Older revisions have that function but it was removed at some point 

https://github.com/HelTecAutomation/ASR650x-Arduino/commit/897da305ed93b18c7aae3f673e9b7d4793c32612#diff-7b33afbac0d26ba044dfcf3bfa70524b672ed70cdfd0d26c72147e45d2118a75R72-R84

aparcar, 2021-01-30:
The user_command was added again in one of the latest commits.

closed 2021-01-30 by aparcar