HelTecAutomation/CubeCell-Arduino #132: No signal on Tx1 pin when CubeCell-GPS is powered from battery

lyusupov, 2020-09-08:
Hardware: **HTCC-AB02S**
Software:  **ASR650x-Arduino V1.1.0**

Test sketch:

```
#include <Wire.h>
#include "cubecell_SSD1306Wire.h"

SSD1306Wire  display(0x3c, 500000, I2C_NUM_0,GEOMETRY_128_64,GPIO10 );

void setup() {
  Serial.begin(38400);
  Serial.println();

  Serial.println("Boot: PASSED");

  pinMode(Vext,OUTPUT);
  digitalWrite(Vext, LOW);

  delay(100);

  display.init();

  display.clear();
  display.setFont(ArialMT_Plain_24);
  display.setTextAlignment(TEXT_ALIGN_CENTER);
  display.drawString(64, 22, "ON");
  display.display();

}

void loop() {

  Serial.println("Test of default UART output");

  delay(100);
}
```

Oscilloscope's probe is connected to Tx1 pad:

[image]

When **the CubeCell is powered over microUSB** - everything is fine:

[image]

However, when the same test is running on the board, **powered by a 3.7V Li-Po battery cell** - there is no proper signal detected on the Tx1 pad:

[image]

If I touch the probe with my finger, the scope indicates that the pad is in high impedance state:

[image]

Rest of HTCC-AB02S hardware (**including Rx2/Tx2** Air530 GNSS pins) is operating nicely from the battery power.

15883893721, 2020-09-09:
hi,

We got the result. This was what we designed for low power consumption.
The situation is roughly like this:
When USB is powered, the serial port chip CP2102 will be powered on, and RX1 will be pulled to a high level . When you use this statement “Serial.begin();” in your code, the level of RX1 will be automatically detected. If RX1 is high, the serial port will be opened, so TX1 can The waveform is detected.

When only battery powered, cp2102 is not poweredand RX1 is not powered on either. The serial port is not started, RX1 has no waveform.

One of the solutions: Before the system starts (only battery power supply), the hardware method is to set RX1 to high level.

lyusupov, 2020-09-09:
Good. Thank you for your explaination! Actually, this is close to that I expected to read.

> This was what we designed for low power consumption.

Have you **intentionally** designed that way ? 😄 
If it is so - I doubt that it was best possible solution...

Designer should be aware that user would like (and have a right) to connect further devices to the UART0 pins, such as Bluetooth/BLE/Wi-Fi or other bridge.

Ok. Could you, please, suggest 'software only' solutions first to fix this issue?
As an example, if I do:

```
pinmode(UART_RX1, OUTPUT);
digitalWrite(UART_RX1, HIGH);
Serial.begin(38400);
```
&nbsp;- this will NOT work as expected, right?

Next one is a 'hardware solution'. If I'll connect **external pullup** resistor to the Rx1 pin - it will work, right ?
If so, should we consider to add the pullup resistor (connected to, say, software-contolled **Vext** power supply) in further revisions of the CubeCell-GPS board ?

lyusupov, 2020-09-09:
Ok, **disregard my first question** - I see your picture attached to the CubeCell forum answer...

lyusupov, 2020-09-09:
I suppose that we should find a way to make somehow [this auto-sensing code](https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/cores/asr650x/Serial/HardwareSerial.cpp#L36) optional... Shouldn't we ?

```
if(digitalRead(UART_RX)==UART_RX_LEVEL)//uart start when uart chip powered
```

15883893721, 2020-09-09:
[image]

Software method： Block the "if judgment" in the picture

15883893721, 2020-09-09:
C:\Users\admin\Documents\Arduino\hardware\CubeCell\ASR650x-Arduino\cores\asr650x\Serial

15883893721, 2020-09-09:
> 
> 
> I suppose that we should find a way to make somehow [this auto-sensing code](https://github.com/HelTecAutomation/ASR650x-Arduino/blob/master/cores/asr650x/Serial/HardwareSerial.cpp#L36) optional... Shouldn't we ?
> 
> ```
> if(digitalRead(UART_RX)==UART_RX_LEVEL)//uart start when uart chip powered
> ```

yes， this is a good idea

closed 2020-09-17 by lyusupov