HelTecAutomation/CubeCell-Arduino #134: LoRaWan, First user packet length vs Data rate max packet size validation does not use user defined Data Rate

leroyle, 2020-09-15:
After a network join:
 
The first user packet length vs data rate allowed max packet length validation is not using the user defined data rate for the validation. Rather the  (for US915) US915_DEFAULT_DATARATE as defined in Region915.h is used.

In our case the US915_DEFAULT_DATARATE in Region915.h is set to DR_3,
the user defined data rate defined in LoRaWan_APP.cpp is set to DR_0 ( or using the new API)
The packets we try to send will be 15 bytes long, exceeding the DR_0 allowed.

In this case if our first packet exceeds the max length allowed by the DR setting this, LoRaWan_APP level, validate will not catch the over size because DR_3 is used during this first validation. Fortunately the lower level,  LoRaMac level, ScheduleTx() catches the problem and aborts the send.

Subsequent sends are caught at the LoRaWan_APP level as the data rate used for the subsequent validates has been updated to the user defined data rate by the time the second send packet length is validated.

It seems even that first validate should fail at the LoRaWan_APP level, but it is not.

Heltec-Aaron-Lee, 2020-09-27:
the user defined data rate defined in LoRaWan_APP.cpp is for Adr disabled

leroyle, 2020-09-27:
The concern expressed above was encountered before any ADR ON testing was done. The latest comment in #135 includes this concern as well.