HelTecAutomation/CubeCell-Arduino #137: New board request.

retrodime, 2020-09-15:
Hello, is it possible to make this work with this board???

https://www.aliexpress.com/item/4000458864812.html?spm=a2g0o.productlist.0.0.178258cbMTkQa4&algo_pvid=2c21f90c-1646-47a2-be95-f4417b98ed5f&algo_expid=2c21f90c-1646-47a2-be95-f4417b98ed5f-6&btsid=0bb0624416001851040415948e108a&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

lyusupov, 2020-09-17:
This GitHub project is not applicable for the T-Solar, because:
- T-Solar uses ASR650**5** (STM8 MCU architecture)
- CubeCell uses ASR650**1/2** (PSoC4+ MCU architecture)

they are completely different.

Heltec-Aaron-Lee, 2020-09-18:
Hi @retrodime, @lyusupov gave the correct explanation. 👍 

We have evaluated whether it is valuable to transplant Arduino from ASR6505, and found that the meaning of the STM8 core is not very big. Later, another new chip of ASR attracted us (because of the NDA contract, we cannot disclose more about it to you for the time being), and we decided to focus on this new chip.