HelTecAutomation/CubeCell-Arduino #145: Frequence plan for AS2

Ldsrc2008, 2020-10-08:
I'm currently have V1.1 (latest one in board manager)on my arduino IDE, it only has AS923 in the region section, I saw you have AS1, AS2 support on the board.txt, how do I update mine to have AS2 support?
I tried to uninstall and reinstall again, but it does not change...

Heltec-Aaron-Lee, 2020-10-09:
Already added in the newest development framework, but the V1.1.0 didn't include that function. Before we generate a new release, please [install the CubeCell development framework via Git.](https://heltec-automation-docs.readthedocs.io/en/latest/cubecell/quick_start.html#via-git)
[image]

Ldsrc2008, 2020-10-12:
I see, then how can I update it in platformIO? it seems the support is through arduino framwork...

Heltec-Aaron-Lee, 2020-10-13:
We will do that in the next few days.