HelTecAutomation/CubeCell-Arduino #184: Air530Z satellite network configuration not working

31warren, 2021-06-22:
The Air530Z library sets different satellite networks but no satellites are received on most settings. Only GPS & GLONAS show any satellites.

Using the old Air530 Library it is possible to see satellites on the other networks - showing that the hardware works ok. 

Also, the number of satellites received by a 530Z board is less than half the number received by a board with the previous Air530 chip. Without a datasheet for the 530Z it is not possible to check if the 530Z has receiver LNA gain settings. An external active antenna is needed with the board - this was not always needed for the Air530.

Max-Plastix, 2022-01-14:
Consider adding 
` GPS.setmode(MODE_GPS_BEIDOU_GLONASS);`
to your code.  With this, I see all three satellite constellations in view, and all are used in the solution.

There is a (Chinese) datasheet for the AIR530Z available.

31warren, 2022-01-14:
Thanks for the reply.

The main issue is that the board with AIR530Z does not receive any satellites unless you add another GPS antenna. With previous versions of the board that used the Air530 satellites could be received using the onboard antenna alone. It would be good if future versions of the board worked with the onboard antenna. ?? compatibility issue between the antenna and the Air530Z or a PCB design problem??

Max-Plastix, 2022-01-14:
The CubeCell I have with AIR530Z has two antenna paths.. one is an on-board ceramic antenna that goes to a (Maxim) RF amplifier and into the 530Z.  The second path is a direct feed to the IPEX/U.Fl connector with 3.3v power for an active antenna.  A larger 25x25mm Active antenna works better than the internal chip antenna + amp, by far.

I've also experimented with removing the ceramic on-board antenna, and even the RF Amp to see if they were adding noise to the active antenna path.  That was inconclusive.

I will say that even with a 10x10mm active antenna, I get solid lock in 10-15 seconds near a window, 13 to 18 sats in view, and participation from all three constellations.

Max-Plastix, 2022-01-14:
(Note that the Heltec schematic has these mislabeled/reversed)

31warren, 2022-01-14:
Thanks again for the information - I am also using an active antenna and getting 10 sats.

It is a pity that this version of the module needs an external antenna - the original version worked very well on its own. 

I would recommend that Heltec consider improving the performance of the onboard antenna reception in the next version.

closed 2022-01-14 by 31warren