HelTecAutomation/CubeCell-Arduino #195: 0.0.7 last working version with LoraWAN and HTCC-AB01

nmaas87, 2021-09-25:
I got 2 HTCC-AB01 868 MHz and my own TTN / Chirpstack Gateway (depending on switching SDcard :) ) - for quite a long time I have been trying to get both modules working, however, like said in #107 - with 1.3.0 and 1.2.0 - most of the time my board only does authenticate with TTN, but does not work further - or lose a lot of packets (never receiving the gateway, even though very close to it - on another table). I went down with the LoraWAN example all versions until i reached stable service.

All the other versions never worked.
How can I fix that?
Are both modules defective and should be given back?

Settings as follows:
CubeCell Version 0.0.7
Region REGION_EU868
Class_A
Netmode: OTAA
ADR: ON
Uplinkmode: Unconfirmed
Net_Res: OFF
AT_SUPPORT: OFF
RGB_DEACT

````
#include <LoRaWan_APP.h>
#include <Arduino.h>

/*
Code from: https://github.com/LukePrior/TTN-BME280-Weather-Station-Heltec-CubeCell-HTCC-AB01

function Decoder(bytes, port) {
  var temperature = bytes[0]<<24>>16 | bytes[1];
  var humidity = (bytes[2] << 8) | bytes[3];
  var pressure = ((bytes[4]) << 24) + ((bytes[5]) << 16) + ((bytes[6]) << 8) + ((bytes[7]));
  var battery = (bytes[8] << 8) | bytes[9];
  var battery_level = bytes[11];
  
  return {
    temperature: temperature / 100,
    humidity: humidity,
    pressure: pressure / 100,
    battery: battery / 1000,
    battery_level: battery_level
  }
}
*/

/* OTAA para*/
uint8_t devEui[] = { REDACTED }; //#Insert you Development EUI here as indivudal bytes replacing each XX pair
uint8_t appEui[] = { REDACTED }; //#Insert you Application EUI here as indivudal bytes replacing each XX pair
uint8_t appKey[] = { REDACTED }; //#Insert you Application Key here as indivudal bytes replacing each XX pair

/* ABP para*/

uint8_t nwkSKey[] = { REDACTED };
uint8_t appSKey[] = { REDACTED };
uint32_t devAddr =  ( uint32_t )REDACTED;

bool ENABLE_SERIAL = true; // enable serial debug output here if required
uint32_t appTxDutyCycle = 100000; // the frequency of readings, in milliseconds(set 100s)

uint16_t userChannelsMask[6]={ 0x00FF,0x0000,0x0000,0x0000,0x0000,0x0000 };
LoRaMacRegion_t loraWanRegion = ACTIVE_REGION;
DeviceClass_t  loraWanClass = LORAWAN_CLASS;
bool overTheAirActivation = LORAWAN_NETMODE;
bool loraWanAdr = LORAWAN_ADR;
bool keepNet = LORAWAN_NET_RESERVE;
bool isTxConfirmed = LORAWAN_UPLINKMODE;
uint8_t appPort = 2;
uint8_t confirmedNbTrials = 4;

int temperature, humidity, batteryVoltage, batteryLevel;
long pressure;

int counter = 0;

static void prepareTxFrame( uint8_t port )
{
  counter = counter + 1;

  temperature = counter;
  humidity = 0;
  pressure = 0;
  
  batteryVoltage = getBatteryVoltage();
  batteryLevel = (BoardGetBatteryLevel() / 254) * 100; 

  appDataSize = 12;
  appData[0] = highByte(temperature);
  appData[1] = lowByte(temperature);

  appData[2] = highByte(humidity);
  appData[3] = lowByte(humidity);

  appData[4] = (byte) ((pressure & 0xFF000000) >> 24 );
  appData[5] = (byte) ((pressure & 0x00FF0000) >> 16 );
  appData[6] = (byte) ((pressure & 0x0000FF00) >> 8  );
  appData[7] = (byte) ((pressure & 0X000000FF)       );

  appData[8] = highByte(batteryVoltage);
  appData[9] = lowByte(batteryVoltage);

  appData[10] = highByte(batteryLevel);
  appData[11] = lowByte(batteryLevel);

  if(ENABLE_SERIAL){
    Serial.print("Temperature: ");
    Serial.print(temperature);
    Serial.print(" mbar, Battery Voltage: ");
    Serial.print(batteryVoltage);
    Serial.print(" mV, Battery Level: ");
    Serial.print(batteryLevel);
    Serial.println(" %");
  }
}

void setup()
{
  
  boardInitMcu();
  if(ENABLE_SERIAL){
    Serial.begin(115200); 
  }
  deviceState = DEVICE_STATE_INIT;
  LoRaWAN.ifskipjoin();

}

void loop()
{
  switch( deviceState )
  {
    case DEVICE_STATE_INIT:
    {
      printDevParam();
      LoRaWAN.init(loraWanClass,loraWanRegion);
      deviceState = DEVICE_STATE_JOIN;
      break;
    }
    case DEVICE_STATE_JOIN:
    {
      LoRaWAN.join();
      break;
    }
    case DEVICE_STATE_SEND:
    {
      prepareTxFrame( appPort );
      LoRaWAN.send();
      deviceState = DEVICE_STATE_CYCLE;
      break;
    }
    case DEVICE_STATE_CYCLE:
    {
      // Schedule next packet transmission
      txDutyCycleTime = appTxDutyCycle + randr( 0, APP_TX_DUTYCYCLE_RND );
      LoRaWAN.cycle(txDutyCycleTime);
      deviceState = DEVICE_STATE_SLEEP;
      break;
    }
    case DEVICE_STATE_SLEEP:
    {
      LoRaWAN.sleep();
      break;
    }
    default:
    {
      deviceState = DEVICE_STATE_INIT;
      break;
    }
  }
}
````

ardadenker, 2022-01-11:
I am having the same issue with my HTCC-AB01 where it only joins the network with no uplink transfer. Have you found any solution or workaround to this?

nmaas87, 2022-01-12:
Hi @ardadenker - no, there is no solution and I had no one at HelTec offering support, sadly. So the best solution is probably to use the outdated framework or using different, non-HelTec products.

1rabbit, 2022-01-12:
Hi, I have a solution and that would be great if you guys can tell me if it works for you as well
https://github.com/1rabbit/CubeCell-Arduino/commit/58cb65dc7ddd5e7e2ceb5ae23858d678207062d4

nmaas87, 2022-01-12:
@1rabbit I did install your changes to my local repo and am going to try this within the next days, but I will need some time as my gateway is currently down, but I'll leave this tab open and gonna let you know if it works for me.

ardadenker, 2022-01-12:
@nmaas87 @1rabbit This fix works on the latest version 1.3,0, changed the preamble size to 8 instead of 16. It almost instantly joined the network and sends uplink on the default LoraWan example script. Thanks @1rabbit, much appreciated for sharing it!

nmaas87, 2022-01-13:
Dear @1rabbit - perfect fix!
Thanks to @ardadenker for already trying it - I hastly got my gateway working again - just to find out that 1.3.0 is - with your patch (using EU_868 :) - just working perfectly. Like. Flawlessly. I really had nights on nights trying - and it did not work out - but this works better than the 0.0.7 version 💯.

So did HelTec did screw-up there? Is the preamble only 8 bit long? Can you get this merged to the CubeCell as PR? :) 

Thanks a lot!