HelTecAutomation/CubeCell-Arduino #196: save eeprom or alternative more than 255 number value  or string

halukmy, 2021-09-27:
what you use for store string or integer for non delete memmory?

ScholliYT, 2021-12-15:
You can easily use the provied `EEPROM.put(address, structobj);` function to save a struct with arbitrary data types like `uint32_t` to the EEPROM. You can use the function with code similar to this:

```cpp
// define struct to save to EEPROM
// you can add any values here
struct MyConfig {
  uint8_t magicByte; // used to check for Config in eeprom
  uint8_t version;
  uint16_t someValue;

  MyConfig() : magicByte(0x42), version(0x03), someValue(0x4242) { }
};

// instantiate a obj of our Struct
MyConfig config = MyConfig();
uint16_t configAddress = 0;

// Save it to EEPROM with this fucntion using EEPROM.put
void saveConfig() {
  EEPROM.put(configAddress, config);
  if (EEPROM.commit()) {
    Serial.println(F("Config successfully committed to EEPROM"));
  } else {
    Serial.println(F("ERROR! Config EEPROM commit failed"));
  }
}

void loadConfig() {
  // get() can be used with custom structures too.
  MyConfig configFromEEPROM;
  EEPROM.get(configAddress, configFromEEPROM);
  
  // Check if there is a saved valid config
  if(configFromEEPROM.magicByte == config.magicByte) {
    if(configFromEEPROM.version == config.version) {
      EEPROM.get(configAddress, config);
    } else {
      Serial.println(F("WARNING Config version mismatch. Will not load config from EEPROM."));
    }
  }
}

```

eikaramba, 2021-12-30:
i used the code from the examples folder https://github.com/HelTecAutomation/CubeCell-Arduino/tree/master/libraries/EEPROM/examples

however i observed that while using the board adresses 0-255 seem to randomly be resetted to 0. i don't know why but using 256-512 seems to be safe and even after powering off the board, the saved values are still readable on the next boot.

ScholliYT, 2021-12-30:
> i used the code from the examples folder https://github.com/HelTecAutomation/CubeCell-Arduino/tree/master/libraries/EEPROM/examples
>
> however i observed that while using the board adresses 0-255 seem [...]

This is strange. I am using address 0 for 10+ boards without that issue. I am using the cubecell module plus.

eikaramba, 2021-12-30:
yeah it baffles me too. ;) i am using the dev board. i checked multiple times it is not me writting the EEPROM. i just now that when it actually is doing something (join and one manual app port uplink) after i disconnect the cable and connect it again the first 0-255 addresses are resetted to 0 (although nothing ever was writing a zero in my code). i only need the whole code termporarly for debugging. so i will just leave it like it is. if anyone else experiences the same behaviour he/she can try this out :)