HelTecAutomation/CubeCell-Arduino #202: Only 84 uploads

RobSmeenk, 2021-11-01:
I am measuring temperatures in my beehives.

Setup: cubecell-AB01, EU868, ABP, ADR=off, UNCONFIRMED uplinks, measurement every 15 minutes, deepsleep in between. My own Things Indoor gateway handles all traffic.

This worked great under TTN V2 for > 1year with hardly any uplinks lost.

Migrated my gateway to TTN V3, everything still worked fine.

Then I reprogrammed my cubecells to make the uplinks to V3 rather than V2.

After initial trouble (had to edit the RegioEU868.h file to get the correct RX timings) it seemed to work.

But:

- all my devices stop after exactly 84 uplinks, after reboot I have noticed that the LIPO batteries are partially drained, minus 0.3 Volt or so. 

- sometimes each uplink causes a downlink.

- sometimes the cycle changes from 15 minutes to 10-15 seconds.

Any suggestions what I can do to resolve these issues?

RobSmeenk, 2021-11-01:
Some more info on the 84 uplinks:
- uplink 84 is also the last uplink seen by my gateway and application.
- battery drain increases, when time to manual reboot increases, so possibly the cubecell is not going into deepsleep after 84 cycles and therefore drains the battery much faster and does not reach its next transmit cycle.
- my arduino code is from https://github.com/HelTecAutomation/CubeCell-Arduino/blob/master/libraries/LoRa/examples/LoRaWAN/LoRaWan/LoRaWan.ino

eikaramba, 2021-11-28:
hey there, i am also using the V3 console. i cannot tell you 100% because i just reseted my device but i am quite sure that before it was doing more uplinks than 84.

i have ADR ON, confirmed uplinks however

may i ask what you did when you said you edit the regioeu868.h file ? for me everythign is working but i just want to make sure i did not misconfigured anything. i am however setting the DR for no adr manually
```
LoRaWAN.init(loraWanClass,loraWanRegion);
LoRaWAN.setDataRateForNoADR(DR_0);
```

maybe post your 1:1 code and i can also have a look.