HelTecAutomation/CubeCell-Arduino #207: std::queue isnt working

miathedev, 2021-12-05:
Did someone manage to get std::queue working?
It wont compile due to a cascade of errors.

```
#include <Arduino.h>
#include <queue>

std::queue <uint16_t> test;
void setup() {
// write your initialization code here
}

void loop() {
// write your code here
}
```
<details>
  <summary>Full log output</summary>

```
====================[ Build | Debug | cubecell_board ]==========================
/home/mime/jetbrains/clion-2021.2/bin/cmake/linux/bin/cmake --build /home/mime/CLionProjects/untitled8/cmake-build-cubecell_board --target Debug -- -j 9
Processing cubecell_board (platform: asrmicro650x; board: cubecell_board; framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/asrmicro650x/cubecell_board.html
PLATFORM: ASR Microelectronics ASR650x (1.3.1) > Heltec CubeCell-Board (HTCC-AB01)
HARDWARE: ASR6501 48MHz, 16KB RAM, 128KB Flash
PACKAGES: 
 - framework-arduinoasrmicro650x 1.3.0 
 - tool-cubecellelftool 0.0.1 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
Compiling .pio/build/cubecell_board/src/main.cpp.o
In file included from /home/mime/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/9.2.1/deque:60,
                 from /home/mime/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/9.2.1/queue:60,
                 from src/main.cpp:2:
/home/mime/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/9.2.1/bits/stl_algobase.h:246:56: error: macro "min" passed 3 arguments, but takes just 2
  246 |     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
      |                                                        ^
In file included from src/main.cpp:1:
/home/mime/.platformio/packages/framework-arduinoasrmicro650x/cores/asr650x/Arduino.h:68: note: macro "min" defined here
   68 | #define min(a, b) ((a)<(b)?(a):(b))
      | 
In file included from /home/mime/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/9.2.1/deque:60,
                 from /home/mime/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/9.2.1/queue:60,
                 from src/main.cpp:2:
/home/mime/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/9.2.1/bits/stl_algobase.h:268:56: error: macro "max" passed 3 arguments, but takes just 2
  268 |     max(const _Tp& __a, const _Tp& __b, _Compare __comp)
      |                                                        ^
In file included from src/main.cpp:1:
/home/mime/.platformio/packages/framework-arduinoasrmicro650x/cores/asr650x/Arduino.h:69: note: macro "max" defined here
   69 | #define max(a, b) ((a)>(b)?(a):(b))
      | 
/home/mime/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/9.2.1/bits/stl_algobase.h:198:5: error: expected unqualified-id before 'const'
  198 |     min(const _Tp& __a, const _Tp& __b)
      |     ^~~
/home/mime/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/9.2.1/bits/stl_algobase.h:198:5: error: expected ')' before 'const'
/home/mime/.platformio/packages/framework-arduinoasrmicro650x/cores/asr650x/Arduino.h:68:20: note: to match this '('
   68 | #define min(a, b) ((a)<(b)?(a):(b))
      |                    ^
/home/mime/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/9.2.1/bits/stl_algobase.h:198:5: error: expected ')' before 'const'
  198 |     min(const _Tp& __a, const _Tp& __b)
      |     ^~~
/home/mime/.platformio/packages/framework-arduinoasrmicro650x/cores/asr650x/Arduino.h:68:19: note: to match this '('
[... 1419 lines cut ...]
set(CMAKE_CXX_COMPILER_WORKS 1)

project("untitled8" C CXX)

include(CMakeListsPrivate.txt)

if(EXISTS ${CMAKE_CURRENT_SOURCE_DIR}/CMakeListsUser.txt)
include(CMakeListsUser.txt)
endif()

add_custom_target(
    Production ALL
    COMMAND platformio -c clion run "$<$<NOT:$<CONFIG:All>>:-e${CMAKE_BUILD_TYPE}>"
    WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
)

add_custom_target(
    Debug ALL
    COMMAND platformio -c clion debug "$<$<NOT:$<CONFIG:All>>:-e${CMAKE_BUILD_TYPE}>"
    WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
)

add_executable(Z_DUMMY_TARGET ${SRC_LIST})

```

benner, 2022-06-07:
I have exact same issue. In my case it's triggered by `#include <list>`

benner, 2022-06-07:
Did some digging. Replaced with newest `gcc-arm-none-eab`i from ARM - does not help. Then looked into other ARM based Ardiuno MCU and compared Arduino.h.  Temporary solve problem by commenting these lines:

https://github.com/HelTecAutomation/CubeCell-Arduino/blob/master/cores/asr650x/Arduino.h#L67-L68

Above there are lines:
```
//using std::abs;                                                                       
//using std::max;                                                                       
//using std::min;
```

I think there was some doubt use std or define and define won :-)

benner, 2022-06-07:
Update: need to uncoment 

```
#include <algorithm>

using std::abs;                                                                       
using std::max;                                                                       
using std::min;
```