HelTecAutomation/CubeCell-Arduino #211: Cubecell 1/2AA (HTCC-AB02A) won't Tx under battery power

aimeeraymond, 2021-12-17:
I have two Cubecell 1/2AA boards. They will transmit LoRaWan to TTN without issue when connected to the USB port (with or without the battery installed).  If I disconnect the node from USB power (hence using battery power), then the node will not transmit.

To isolate the issue, I installed the Lora Ping-Pong application onto both nodes using, https://github.com/HelTecAutomation/CubeCell-Arduino/tree/master/libraries/LoRa/examples/LoRaBasic/pingpong.  I commented out the lines for "turnOnRGB()", as this board does not have the RGB component and the script won't compile without doing so.
The ping-pong script runs for both nodes when they are powered by USB.  When I unplug one node from USB, so it is running on battery, the ping-pong immediately stops.  The USB-connected node's serial output indicates that is it waiting to receive a packet (hence the Battery-powered node failed to transmit).   Putting the Battery-powered node back on USB power, the ping-pong immediately resumes, but it's packet counter resets to 1.

When on USB power, there is a red-LED on between the USB connector and the RST button.
When on battery power, there are no LED's on.

On one of the nodes, the first time I installed the battery there was a flicker of a green LED located between the USB connector and the USER button.  I have not been able to get the green LED on after that first flicker.  On the second node, I have never seen the green LED on.

For Node A, the battery readings (taken by multimeter) are:
   loose battery,  3.55V, 
    no USB connection, installed in the battery holder reads 3.44V and alternates to 2.79V (goes repeatedly back and forth at the pace of the ping-pong)
          VDD & Pin 19 =3.4/2.79 alternating, VIN & VEXT = 0
    with USB connection, installed in battery holder reads solid 3.55V and is increasing slowly (charging?), not alternating, stopped increasing at 3.65V

For Node B,
   loose battery, 3.44V
    no USB connection, in battery holder, 1.60V not alternating like Node A
      VDD & Pin 19 =1.6V not alternating, VIN & VEXT = 0
     with USB connection, in battery holder, 3.44 V and increased to 3.6V, not alternating

Given the difference between the battery voltage when in the battery holder, and one alternating vs not alternating, I put the same ping-pong script on a third CubeCell Dev Board (AB01) running on USB power.  Both Node A & Node B resumed ping-pong when they were USB-powered.  

CubeCell Dev Board on battery power would ping-pong with 1/2AA Node A or Node B, but only when the 1/2AA node was on USB-power.

Why won't CubeCell 1/2AA transmit under battery power?

jezdd, 2022-01-03:
Share your full code, we have 12 x CubeCell 1/2AA working on battery - last for years too.

aimeeraymond, 2022-01-03:
I am using the example PingPong script.  I commented out Line 135 "// turnOnRGB(0,0);".  I installed same script on two CubeCell 1/2AA.

`/* Heltec Automation Ping Pong communication test example
 *
 * Function:
 * 1. Ping Pong communication in two CubeCell device.
 * 
 * Description:
 * 1. Only hardware layer communicate, no LoRaWAN protocol support;
 * 2. Download the same code into two CubeCell devices, then they will begin Ping Pong test each other;
 * 3. This example is for CubeCell hardware basic test.
 *
 * HelTec AutoMation, Chengdu, China
 * 成都惠利特自动化科技有限公司
 * www.heltec.org
 *
 * this project also realess in GitHub:
 * https://github.com/HelTecAutomation/ASR650x-Arduino
 * */

#include "LoRaWan_APP.h"
#include "Arduino.h"

/*
 * set LoraWan_RGB to 1,the RGB active in loraWan
 * RGB red means sending;
 * RGB green means received done;
 */
#ifndef LoraWan_RGB
#define LoraWan_RGB 0
#endif

#define RF_FREQUENCY                                868000000 // Hz

#define TX_OUTPUT_POWER                             5        // dBm

#define LORA_BANDWIDTH                              0         // [0: 125 kHz,
                                                              //  1: 250 kHz,
                                                              //  2: 500 kHz,
                                                              //  3: Reserved]
#define LORA_SPREADING_FACTOR                       7         // [SF7..SF12]
#define LORA_CODINGRATE                             1         // [1: 4/5,
                                                              //  2: 4/6,
                                                              //  3: 4/7,
                                                              //  4: 4/8]
#define LORA_PREAMBLE_LENGTH                        8         // Same for Tx and Rx
#define LORA_SYMBOL_TIMEOUT                         0         // Symbols
#define LORA_FIX_LENGTH_PAYLOAD_ON                  false
#define LORA_IQ_INVERSION_ON                        false

#define RX_TIMEOUT_VALUE                            1000
#define BUFFER_SIZE                                 30 // Define the payload size here

char txpacket[BUFFER_SIZE];
char rxpacket[BUFFER_SIZE];

static RadioEvents_t RadioEvents;
void OnTxDone( void );
void OnTxTimeout( void );
void OnRxDone( uint8_t *payload, uint16_t size, int16_t rssi, int8_t snr );

typedef enum
{
    LOWPOWER,
    RX,
    TX
}States_t;

int16_t txNumber;
States_t state;
bool sleepMode = false;
int16_t Rssi,rxSize;

void setup() {
    Serial.begin(115200);

    txNumber=0;
    Rssi=0;

    RadioEvents.TxDone = OnTxDone;
    RadioEvents.TxTimeout = OnTxTimeout;
    RadioEvents.RxDone = OnRxDone;

    Radio.Init( &RadioEvents );
    Radio.SetChannel( RF_FREQUENCY );
    Radio.SetTxConfig( MODEM_LORA, TX_OUTPUT_POWER, 0, LORA_BANDWIDTH,
                                   LORA_SPREADING_FACTOR, LORA_CODINGRATE,
                                   LORA_PREAMBLE_LENGTH, LORA_FIX_LENGTH_PAYLOAD_ON,
                                   true, 0, 0, LORA_IQ_INVERSION_ON, 3000 );

    Radio.SetRxConfig( MODEM_LORA, LORA_BANDWIDTH, LORA_SPREADING_FACTOR,
                                   LORA_CODINGRATE, 0, LORA_PREAMBLE_LENGTH,
                                   LORA_SYMBOL_TIMEOUT, LORA_FIX_LENGTH_PAYLOAD_ON,
                                   0, true, 0, 0, LORA_IQ_INVERSION_ON, true );
    state=TX;
}

void loop()
{
  switch(state)
  {
    case TX:
      delay(1000);
      txNumber++;
        sprintf(txpacket,"%s","hello");
        sprintf(txpacket+strlen(txpacket),"%d",txNumber);
        sprintf(txpacket+strlen(txpacket),"%s"," Rssi : ");
        sprintf(txpacket+strlen(txpacket),"%d",Rssi);
 //       turnOnRGB(COLOR_SEND,0);

        Serial.printf("\r\nsending packet \"%s\" , length %d\r\n",txpacket, strlen(txpacket));

        Radio.Send( (uint8_t *)txpacket, strlen(txpacket) );
        state=LOWPOWER;
        break;
    case RX:
      Serial.println("into RX mode");
        Radio.Rx( 0 );
        state=LOWPOWER;
        break;
    case LOWPOWER:
      lowPowerHandler();
        break;
        default:
            break;
  }
    Radio.IrqProcess( );
}

void OnTxDone( void )
{
  Serial.print("TX done......");
 // turnOnRGB(0,0);
  state=RX;
}

void OnTxTimeout( void )
{
    Radio.Sleep( );
    Serial.print("TX Timeout......");
    state=TX;
}
void OnRxDone( uint8_t *payload, uint16_t size, int16_t rssi, int8_t snr )
{
    Rssi=rssi;
    rxSize=size;
    memcpy(rxpacket, payload, size );
    rxpacket[size]='\0';
 //   turnOnRGB(COLOR_RECEIVED,0);
    Radio.Sleep( );

    Serial.printf("\r\nreceived packet \"%s\" with Rssi %d , length %d\r\n",rxpacket,Rssi,rxSize);
    Serial.println("wait to send next packet");

    state=TX;
}`

tekk, 2022-02-06:
What type of battery have you used?
Link to exact model?

aimeeraymond, 2022-02-07:
I am using EEMB, Model ER14250, Lithium Thionyl Chloride Battery, 3.6V, ER14250 lithium battery, 3.6V- EEMB Battery <https://www.eemb.com/model/er14250.html> 

 

-aimee

dl5dla, 2022-09-08:
I run into same or similar problem (I guess). Two of my four modules, powered by battery only, join the TTN network and send out regularly messages. The other two do this only when connected to USB. All of them have the same battery type and volatgae across batteries is around 3.6V.

@aimeeraymond: did you solve the problem meanwhile?

aimeeraymond, 2022-09-08:
Peter,

 

I have not solved this problem, and these boards still sit on my desk.  We have successful developed our own PCB using the CubeCell Module Plus, and built nearly 100 nodes.  We have many nodes that do not perform correctly.  I ordered a serious microscope to do a more detailed visual inspection of the boards. We removed the protective cover on the CubeCell ModulePlus.  On inspection, we found potential soldering errors on the CubeCell part.  I’ve not finished the investigation yet.  I do intend to share the pictures with Heltec and to discuss with them.  

 

So from this learning, I would encourage you to get the unit under a microscope and take a look.  If you don’t have a microscope, I can give you recommendation.

 

-aimee

dl5dla, 2022-09-10:
Thanks for your comments. I also assume a physical fault on my Cubecell 1/2 boards. I had one of them operated outside (in waterproof case), it worked for several weeks, but then stopped working. A second one is working on the roof since March wihout problems, but the last two one show the faulty behaviour (as you also initially described.

dl5dla, 2022-10-15:
I did some more investigations with my four modules. Again, I found  that one of the modules stopped working after a longer time although I measured a voltage of ca. 3.4V across the battery (plugged into the holder of the module). When I replaced the battery with a new one, the module starts sending again.

In my opinion all my modules are ok. When not TX'ing, the power consumption of the module is that low, that even a discharged battery (with a big internal resistence in this state) provides a sufficient voltage to the module. But when it's TX'ing, the module draws relevant more current from the battery and this leads to a big voltage across the internal resistence, and therefore  leads to only a small supply voltage below the voltage limits of the module. And one cannot realy report this reduced voltage by getBatteryVoltage() during operation, as this function call is done outside the TX phase.

BTW, I used one of the modules to measure the solar power of a small photovoltaik module on the roof. The battery lasted 6-7 months.