HelTecAutomation/CubeCell-Arduino #213: LoRaWAN not working properly with some Helium hotspots after Version 1.0.0

1rabbit, 2021-12-26:
I had about 95% packet loss with specifically Bobcat Helium Hotspots when using my htcc-ab02s with the last libs (1.1.0, 1.2.0, 1.3.0)

I get 0% packet loss when I switch back to Version 1.0.0

What is the difference on the LoRaWAN between 1.0.0 and 1.1.0, 1.2.0, 1.3.0, is it possible to find a solution without having to downgrade, as I cant run https://github.com/hkicko/CubeCell-GPS-Helium-Mapper on the older version ?

Thank you

nmaas87, 2022-01-12:
Probably related to my findings: https://github.com/HelTecAutomation/CubeCell-Arduino/issues/195

1rabbit, 2022-01-12:
> Probably related to my findings: #195

After several weeks of debugging I finally found a solution : https://github.com/1rabbit/CubeCell-Arduino/commit/58cb65dc7ddd5e7e2ceb5ae23858d678207062d4

nmaas87, 2022-01-13:
Tested @1rabbit solution in #195 - works perfectly. After this, 868 MHz finally seems to work at all for LoRaWAN