HelTecAutomation/CubeCell-Arduino #230: GPS example sketch stuck after a couple of iterations

mehow, 2022-03-29:
Board: CubeCell GPS-6502
Version: 1.4.0
Example: OnBoardGPS > PeriodicLowPowerMode

The sketch gets stuck on "GPS Searching..." after a few iterations. More precisely it gets stuck in `GPS.begin()` while checking for the current baudrate:

```
23:49:00.892 -> GPS Current baudrate detected:9600
23:49:01.004 -> GPS baudrate updating to 9600
23:49:01.040 -> GPS baudrate updated to 9600
23:50:01.975 -> GPS search timeout.
23:50:04.002 -> 2 seconds later into lowpower mode....
23:51:06.118 -> GPS Searching...
23:51:06.118 -> GPS Current baudrate detecting...
23:51:06.336 -> GPS Current baudrate detected:9600
23:51:06.444 -> GPS baudrate updating to 9600
23:51:06.520 -> GPS baudrate updated to 9600
23:51:17.425 -> GPS search timeout.
23:51:19.456 -> 2 seconds later into lowpower mode....
23:52:21.566 -> GPS Searching...
23:52:21.566 -> GPS Current baudrate detecting...
23:52:21.774 -> GPS Current baudrate detected:9600
23:52:21.879 -> GPS baudrate updating to 9600
23:52:21.953 -> GPS baudrate updated to 9600
23:52:32.880 -> GPS search timeout.
23:52:34.901 -> 2 seconds later into lowpower mode....
23:53:37.016 -> GPS Searching...
23:53:37.016 -> GPS Current baudrate detecting...
23:53:37.232 -> GPS Current baudrate detected:9600
23:53:37.356 -> GPS baudrate updating to 9600
23:53:37.389 -> GPS baudrate updated to 9600
23:53:48.333 -> GPS search timeout.
23:53:50.365 -> 2 seconds later into lowpower mode....
23:54:52.474 -> GPS Searching...
23:54:52.474 -> GPS Current baudrate detecting...
23:54:52.702 -> GPS Current baudrate detected:9600
23:54:52.786 -> GPS baudrate updating to 9600
23:54:52.870 -> GPS baudrate updated to 9600
23:55:03.803 -> GPS search timeout.
23:55:05.813 -> 2 seconds later into lowpower mode....
23:56:07.910 -> GPS Searching...
23:56:07.910 -> GPS Current baudrate detecting...
```

(there is no more output after that line)

After adding some print statements I learned that the execution gets stuck inside the `getNMEA()` method if `GPSSerial.available()` continues to return false. This code:

```
String Air530ZClass::getNMEA()
{
	uint32_t starttime = millis();
	String nmea = "";
	char buff[128]; 
	int index = 0;
	while(millis() - starttime <1000)
	{
		Serial.print(".");
		if(GPSSerial.available())
		{
			char c = GPSSerial.read();
			Serial.println(c);
			if(c=='$')
			{
				nmea += c;
				index = GPSSerial.readBytesUntil('\r',buff,127);
				buff[index]=0;
				if(buff[index-3]!='*')
					return "0";

				nmea += (String)buff;
				return nmea;
			}
		}
		Serial.print("_");
	}
	Serial.println("XXX");
	return "0";
}
```

ends up printing

```
00:56:20.923 -> GPS Searching...
00:56:20.959 -> GPS Current baudrate detecting...
00:56:20.959 -> ._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._._.
```

before getting stuck. Notice that it never exits the loop and instead appears crash or get stuck inside `GPSSerial.available()`.

I checked and the problem is not present in 1.3.0 which makes me suspect [this commit](https://github.com/HelTecAutomation/CubeCell-Arduino/commit/09409f086dc887ec8ececee678d9d8254bbfa8a3).

rodri16, 2022-08-22:
Same problem here...

MathieuB1, 2022-09-04:
On my side I have the same issue with lowPowerHandler();. I found that millis() is the function that caused this freeze. What I did is that I removed all the GPS code they provide and get stuck to 9600 baud speed...

https://github.com/MathieuB1/KOREK-Lora-GPS_tracker_CubeCell_GPS-6502/blob/main/GPS_Air530Z1.cpp
https://github.com/MathieuB1/KOREK-Lora-GPS_tracker_CubeCell_GPS-6502/blob/main/GPS_Air530Z1.h

By using those files instead of GPS_Air530Z.h GPS is not stuck anymore!

But in my case I got now stuck on sending the lora message,  stuck on Radio.Send(). For the moment I use the watchdog to reset the chip in case of lora stuck with innerWdtEnable(false);  CySysWdtEnable(); CySysWdtDisable() and feedInnerWdt();

olman011, 2023-06-16:
In case anyone is still having this problem, in PlatformIO the solution is to change the platform in the platformio.ini file.
If you choose the Cubecell GPS board when starting a project it will default to "platform = asrmicro650x". This platform is deprecated and must be manually changed to "platform = heltec-cubecell". 
Once this is done the GPS functionality will work correctly.