HelTecAutomation/CubeCell-Arduino #234: Example "LoRaWan_OnBoardGPS_Air530.ino" stops working after some time

schroepfer, 2022-05-17:
This example from [https://github.com/HelTecAutomation/CubeCell-Arduino/tree/master/libraries/LoRa/examples/LoRaWAN/LoRaWAN_Sensors/LoRaWan_OnBoardGPS_Air530](url)
stops after ca. 5 uplinks to TTN.

rodri16, 2022-08-22:
Same problem here....

505222621, 2022-08-25:
yes,GPS.begin -> serial-begin : malloc(buf),no free(buf),change to global variable or serial-end :free(buf)

olman011, 2023-06-16:
In case anyone is still having this problem, in PlatformIO the solution is to change the platform in the platformio.ini file.
If you choose the Cubecell GPS board when starting a project it will default to "platform = asrmicro650x". This platform is deprecated and must be manually changed to "platform = heltec-cubecell". 
Once this is done the GPS functionality will work correctly.

schroepfer, 2023-06-16:
Thanks!
I'm not so confirm in PlatformIO.
I don't know, where to define the build-flags:

`/*LoraWan region, select in arduino IDE tools*/
LoRaMacRegion_t loraWanRegion = ACTIVE_REGION;
/*LoraWan Class, Class A and Class C are supported*/
DeviceClass_t  loraWanClass = LORAWAN_CLASS;
/*the application data transmission duty cycle.  value in [ms].*/
uint32_t appTxDutyCycle = 180 * 1000;  /* Should be 180 seconds */
/*OTAA or ABP*/
bool overTheAirActivation = LORAWAN_NETMODE;
/*ADR enable*/
bool loraWanAdr = LORAWAN_ADR;
/* set LORAWAN_Net_Reserve ON, the node could save the network info to flash, when node reset not need to join again */
bool keepNet = LORAWAN_NET_RESERVE;
/* Indicates if the node is sending confirmed or unconfirmed messages */
bool isTxConfirmed = LORAWAN_UPLINKMODE;`

Please can you post your platform.ini ?

olman011, 2023-06-21:
This is all I have in my platformio.ini

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:cubecell_gps]
platform = asrmicro650x
board = cubecell_gps
monitor_speed = 115200
framework = arduino

schroepfer, 2023-06-22:
I found it.
Without this build flags PlatformIO can not compile.

Now the software compiles, but the cubecell only runs once and then stops.
:-(

`[env:cubecell_gps]
platform = heltec-cubecell
board = cubecell_gps
framework = arduino

board_build.arduino.lorawan.region = EU868
board_build.arduino.lorawan.adr = ON
board_build.arduino.lorawan.debug_level = FREQ_AND_DIO
board_build.arduino.lorawan.netmode = OTAA
board_build.arduino.lorawan.rgb = ACTIVE
`