HelTecAutomation/CubeCell-Arduino #237: ADC battery value drops after a period

jezdd, 2022-06-07:
Something very frustrating happening around analogueRead/analogueReadmv on ADC on HTCC-AB02A - it work great for a short period but then drops in value for no apparent reason, starts off very accurate and then drops but about 0.5v - I understand how the mv calculations are made, at first I thought it was.

This is when under battery power, all works fine and then for no apparent reason the voltage drops by about 0.5v and continues and the new offset - if the device is reset the ADC reading restore and the process starts again.

Mr-HaleYa, 2022-08-01:
post you code please

jezdd, 2022-08-02:
Will do, let me pull it together

jezdd, 2022-08-30:
Battery ADC does not hold true values, works for a short while but then the results drop by about 0.5v, every now and again it given an accurate result but then goes back to the lower voltage.

This issues only shows up when its run on battery, not via USB.

Payload code below, all the rest is standard from Lorawan examples:

/* Prepares the payload of the frame */
static void prepareTxFrame( uint8_t port )
{
 	pinMode(VBAT_ADC_CTL,OUTPUT);
  	digitalWrite(VBAT_ADC_CTL,LOW);
 	delay(50); 

	uint16_t BatteryVoltage = getBatteryVoltage();
	Serial.print("Battery mV: ");
	Serial.println(BatteryVoltage);
//	Serial.println(BatteryMV);
    appDataSize = 	2;
  	appData[0] = (uint8_t)(BatteryVoltage>>8);
  	appData[1] = (uint8_t)BatteryVoltage;
	
	pinMode(VBAT_ADC_CTL,INPUT);
	Serial.println();
	Serial.flush();
}