HelTecAutomation/CubeCell-Arduino #238: How to read the internal temperature?

tkrahn, 2022-06-07:
According to the [PSoC 4100 family datasheet](https://resource.heltec.cn/download/CubeCell/PSOC_4_PSOC_4100_FAMILY_DATASHEET_PROGRAMMABLE_SYSTEM-ON-CHIP.pdf) the micro controller has an internal temperature sensor.

Page 7:
_Temperature Sensor
PSoC 4100 has one on-chip temperature sensor This consists
of a diode, which is biased by a current source that can be
disabled to save power. The temperature sensor is connected to the ADC, which digitizes the reading and produces a temperature value using Cypress supplied software that includes calibration and linearization._

Is it in any way possible to read it out through the ADC? If so, is there some sample code how to do this?

The reason why I need this is for an extremely light GPS tracker for a small unmanned balloon. We try to save every micro-gram of weight, but temperature would be an important telemetry reading. Therefore external sensors are not an option.

Mr-HaleYa, 2022-07-28:
They need to add the DieTemp.c and DieTemp.h files into the cores/asr650x/projects/PSoC4/ folder 
```c
/*******************************************************************************
* File Name: DieTemp.c
* Version 1.0
*
* Description:
*  This file provides the source code of APIs for the DieTemp_P4 component.
*
*******************************************************************************
* Copyright 2013, Cypress Semiconductor Corporation.  All rights reserved.
* You may use this file only in accordance with the license, terms, conditions,
* disclaimers, and limitations in the end user license agreement accompanying
* the software package with which this file was provided.
*******************************************************************************/

#include "DieTemp.h"

/*******************************************************************************
* Function Name: DieTemp_CountsTo_Celsius
********************************************************************************
*
* Summary:
*  Converts the ADC output to degrees Celsius.
*
* Parameters:
*  int32 adcCounts:
*   Result from the ADC Conversion
*
* Return:
*  Temperature in degrees Celsius
*
* Global variables:
*  None
*
*******************************************************************************/
int32 DieTemp_CountsTo_Celsius(int32 adcCounts)
{
    int32 tempCelsius;
    int32 tInitial;
    int32 tAdjust;
    int32 offsetReg;
    int32 multReg;

    offsetReg = (int16)DieTemp_SAR_TEMP_OFFSET_REG;
    multReg   = (int16)DieTemp_SAR_TEMP_MULTIPLIER_REG;
    /* Calculate tInitial in Q16.16 */
    tInitial = (adcCounts * multReg) + (offsetReg * DieTemp_SAR_TEMP_OFFSET_MULT);

    if(tInitial >= DieTemp_DUAL_SLOPE_CORRECTION)
    {
        /* Shift (100 - tInitial) by 4 bits to prevent scale-adjustment from overflowing. */
        /* Then divide by the integer bits of (100 - cutoff) to end up with a Q16.16 tAdjust */
        tAdjust = (DieTemp_SCALE_ADJUSTMENT * (((int32)DieTemp_HIGH_TEMPERATURE - tInitial)
            / (int32)DieTemp_SCALE_ADJUSTMENT_DIVIDER)) /
            (((int32)DieTemp_HIGH_TEMPERATURE - (int32)DieTemp_DUAL_SLOPE_CORRECTION) /
            DieTemp_SAR_TEMP_DIVIDER);
    }
    else
    {
        /* Shift (40 + tInitial) by 4 bits to prevent scale-adjustment from overflowing. */
        /* Then divide by the integer bits of (40 + cutoff) to end up with a Q16.16 tAdjust */
        tAdjust = ((int32)DieTemp_SCALE_ADJUSTMENT * (((int32)DieTemp_LOW_TEMPERATURE + tInitial)
           / (int32)DieTemp_SCALE_ADJUSTMENT_DIVIDER)) /
            (((int32)DieTemp_LOW_TEMPERATURE + (int32)DieTemp_DUAL_SLOPE_CORRECTION) /
            (int32)DieTemp_SAR_TEMP_DIVIDER);
    }

    /* Add tInitial + tAdjust + 0.5 to round to nearest int. Shift off frac bits, and return. */
    tempCelsius = tInitial + tAdjust + DieTemp_HALF_OF_ONE;

    return (tempCelsius / DieTemp_SAR_TEMP_OFFSET_DIVIDER);
}

/* [] END OF FILE */
```

```c
/*******************************************************************************
* File Name: DieTemp.h
* Version 1.0
*
* Description:
*  This file provides constants and parameter values for the DieTemp_P4
*  component.
*
********************************************************************************
* Copyright 2013, Cypress Semiconductor Corporation.  All rights reserved.
* You may use this file only in accordance with the license, terms, conditions,
* disclaimers, and limitations in the end user license agreement accompanying
* the software package with which this file was provided.
*******************************************************************************/

#if !defined(CY_DIE_TEMP_DieTemp_H)
#define CY_DIE_TEMP_DieTemp_H

#include "cytypes.h"
#include "cyfitter.h"
#include "CyLib.h"
#include "cydevice_trm.h"

/***************************************
*        API constants
***************************************/

#define DieTemp_SAR_TEMP_MULTIPLIER_REG    (*(reg16 *) CYREG_SFLASH_SAR_TEMP_MULTIPLIER)
#define DieTemp_SAR_TEMP_MULTIPLIER_PTR    ( (reg16 *) CYREG_SFLASH_SAR_TEMP_MULTIPLIER)
#define DieTemp_SAR_TEMP_OFFSET_REG        (*(reg16 *) CYREG_SFLASH_SAR_TEMP_OFFSET)
#define DieTemp_SAR_TEMP_OFFSET_PTR        ( (reg16 *) CYREG_SFLASH_SAR_TEMP_OFFSET)

#define DieTemp_SAR_TEMP_OFFSET_SHIFT      (10u)
#define DieTemp_SAR_TEMP_OFFSET_MULT       (0x400)
#define DieTemp_SAR_TEMP_OFFSET_DIVIDER    (0x10000)
#define DieTemp_SAR_TEMP_SHIFT             (16u)
#define DieTemp_SAR_TEMP_DIVIDER           (0x10000)
#define DieTemp_SCALE_ADJUSTMENT_DIVIDER   (16u)
#define DieTemp_HALF_OF_ONE                ((int32)1u << (DieTemp_SAR_TEMP_SHIFT - 1u))

/* (effectively 0.5 << 4u) 0.5 in Q28.4 format */
#define DieTemp_SCALE_ADJUSTMENT           (8)
/* 15 in Q16.16 format */
#define DieTemp_DUAL_SLOPE_CORRECTION      (0xF0000)
/* 100 in Q16.16 format */
#define DieTemp_HIGH_TEMPERATURE           (0x640000)
/* 40 in Q16.16 format */
#define DieTemp_LOW_TEMPERATURE            (0x280000)

/***************************************
*        Function Prototypes
***************************************/

int32 DieTemp_CountsTo_Celsius(int32 adcCounts);

#endif /* CY_DIE_TEMP_DieTemp_H */

/* [] END OF FILE */
```
then since its using the ASR probably some code in cores/asr650x/cores/ASR_Arduino.h and a corresponding .c file
not sure what else they would need to do but it _Can_ be added since like you said the SOC supports it
I would personally love to see it since I wouldn't mind an onboard temp sensor even if its technically a die sensor

you could even try doing it yourself with that code and some documentation reading 🤷‍♂️ 🙃

tkrahn, 2022-07-28:
Thanks Mr-HaleYa,
Your code nicely shows how to calculate the temperature. This is very useful, once you have the adcCounts.
However what I'm trying to figure out is how to get the adcCounts for the Die Temperature Sensor in the first place.

I've found from  https://community.infineon.com/t5/Knowledge-Base-Articles/Calculate-Die-Temperature-with-DieTemp-Component-in-PSoC-4-KBA229527/ta-p/260201
the following code:

```
int32 u32 DieTempGet(void) {
	int16 adc_count;
	uint32 u32_regTemp;
	int32 Celsius;
	u32_regTemp = CY_GET_REG32(0x403A0000u);

	CY_SET_REG32(0x403A0000u, ((u32_regTemp & 0xFFFFFF00u) | 0x00000040u)); // Before DieTemp measurement, change ADC Vref connection to internal Vref(1.2-V)
	ADC_EnableInjection(); // INJ Channel Enable
	ADC_StartConvert();    // ADC Convert start
	while(0u == ADC_IsEndConversion(ADC_RETURN_STATUS_INJ)){}
	adc_count = ADC_GetResult16((uint32)INJ_CHN);

	// DO NOT need adc count adjusted/calibrated, because Vref has been changed to internal vref (1.2-V)
	Celsius = DieTemp_CountsTo_Celsius(adc_count); // get temp celsius Degree
	CY_SET_REG32(0x403A0000u, u32_regTemp); // after measurement, restore original ADC vref connection register
	
	return Celsius;
 }
```

doesn't compile because ADC_EnableInjection() and ADC_StartConvert() are not declared.

I was trying the following:
```
/*  Trying to get the ADC value of  internal temperature sensor of PSoC 4100 */
uint16_t getOnChipTempAdc(void)
{
	float temp = 0;
	uint16_t volt;
	uint8_t pin = ADC2; // internal temperature sensor of PSoC 4100 ????

	pinMode(VBAT_ADC_CTL,OUTPUT);
	digitalWrite(VBAT_ADC_CTL,LOW);

	for(int i=0;i<50;i++) //read 50 times and get average
		temp+=analogReadmV(pin);
	volt = 2 * temp / 50;
	

	pinMode(VBAT_ADC_CTL, INPUT);

	// volt = volt*2;
	return volt;
}

/*  Get the ADC value in mV of different ADC Pins (ADC2, ADC3) */
uint16_t getAdcMilliVolt(uint16_t pin)
{
	float temp = 0;
	uint16_t volt;
	
	pinMode(VBAT_ADC_CTL,OUTPUT);
	digitalWrite(VBAT_ADC_CTL,LOW);

	for(int i=0;i<50;i++) //read 50 times and get average
		temp+=analogReadmV(pin);
	volt = 2 * temp / 50;
	

	pinMode(VBAT_ADC_CTL, INPUT);

	// volt = volt*2;
	return volt;
}

```
but I'm not sure if I'm reading the correct ADC "pin" and if the power on the internal diode is enabled correctly.
The adc count doesn't seem to reflect anything related to a temperature.

Mr-HaleYa, 2022-07-28:
oh, I see. looking at this repos code it doesn't look like they use the internal ADC at all... so you would need the files to enable and control it
Those functions you listed need 2 files in cores/asr650x/projects/PSoC4/

```c
/*******************************************************************************
* File Name: ADC.h
* Version 2.50
*
* Description:
*  This file contains the function prototypes and constants used in
*  the Sequencing Successive Approximation ADC Component.
*
* Note:
*
********************************************************************************
* Copyright 2008-2017, Cypress Semiconductor Corporation.  All rights reserved.
* You may use this file only in accordance with the license, terms, conditions,
* disclaimers, and limitations in the end user license agreement accompanying
* the software package with which this file was provided.
*******************************************************************************/

#if !defined(CY_ADC_SAR_SEQ_ADC_H)
#define CY_ADC_SAR_SEQ_ADC_H

#include "cytypes.h"
#include "CyLib.h"

/***************************************
*      Data Struct Definition
***************************************/

/* Sleep Mode API Support */
typedef struct
{
    uint8 enableState;
    uint32 dftRegVal;
} ADC_BACKUP_STRUCT;

/**************************************
*    Enumerated Types and Parameters
**************************************/

/*  Clock Source setting constants */
#define ADC__EXTERNAL 0
#define ADC__INTERNAL 1

/*  Sample Mode setting constants */
#define ADC__FREERUNNING 0
#define ADC__HARDWARESOC 1

/*  Reference type setting constants */
#define ADC__VDDA_2 0
#define ADC__VDDA 1
#define ADC__INTERNAL1024 2
#define ADC__INTERNAL1024BYPASSED 3
#define ADC__INTERNALVREF 4
#define ADC__INTERNALVREFBYPASSED 5
#define ADC__VDDA_2BYPASSED 6
#define ADC__EXTERNALVREF 7

/* Input buffer gain setting constants */
#define ADC__DISABLED 0
#define ADC__ONE 1
#define ADC__TWO 2
#define ADC__FOUR 3
#define ADC__EIGHT 4
#define ADC__SIXTEEN 5

/* Negative input setting sonstants in single ended mode */
#define ADC__VSS 0
[... 1474 lines cut ...]
        /* Halt CPU in debug mode if channel is out of valid range */
        CYASSERT(chan < ADC_TOTAL_CHANNELS_NUM);

        /* Divide the adcCount when accumulate averaging mode selected */
        #if(ADC_DEFAULT_AVG_MODE == ADC__ACCUMULATE)
            if((ADC_channelsConfig[chan] & ADC_AVERAGING_EN) != 0u)
            {
                adcCounts /= ADC_DEFAULT_AVG_SAMPLES_DIV;
            }
        #endif /* ADC_DEFAULT_AVG_MODE == ADC__ACCUMULATE */

        /* Subtract ADC offset */
        adcCounts -= ADC_offset[chan];

        volts = ((float32)adcCounts * ADC_10V_COUNTS) / (float32)ADC_countsPer10Volt[chan];

        return( volts );
    }

#endif /* End ADC_DEFAULT_JUSTIFICATION_SEL == ADC__RIGHT */

/* [] END OF FILE */
```

Big files.... let me know what adding those 4 files does

tkrahn, 2022-07-28:
Holy guacamole! 1594 lines of code, just to read the temperature :)
I'm a little worried that this will completely exhaust the memory resources of the small micro controller and we may have conflicts with the existing ADC code. Yet most of it are definitions of constants that hopefully get eliminated during compiler optimizations. Therefore I'm going to try it.

Mr-HaleYa, 2022-07-29:
Lol yes, that is a ton of code! It should not exahst the SoC as it is the code that the manufacturer recommends, and It looks like 90% of it is just declarations. 
Let me know how it goes 😃

sysfu, 2023-09-12:
> Holy guacamole! 1594 lines of code, just to read the temperature :) I'm a little worried that this will completely exhaust the memory resources of the small micro controller and we may have [...]

Were you successful in getting your code to compile and yield live temperature readings? I'm thinking about doing something similar with a Heltec CubeCell-GPS board.