HelTecAutomation/CubeCell-Arduino #246: Software Reboot?

Mr-HaleYa, 2022-07-29:
Is there any way to trigger a software reboot? I don't see any code for it after reading through all the examples and almost every file in this repo. 

My current way of triggering a software reboot is to enable the watchdog timer and then cause a freeze so the timer lapses and the watchdog reboots the device. 😅

ScholliYT, 2022-07-31:
I also use the watchdog timer method for sofware reboot. Do you see any problem with that method?

Mr-HaleYa, 2022-07-31:
I mean if it works, it works. It's just a little hacky and my only concern is if I ever want to use the device without the Watchdog then I'm not sure what to do.

Glad it's not just me doing it that way 😅

tuxphone, 2022-07-31:
Hello, please try this:

#include "CyLib.h"

To reboot, use:   CySoftwareReset();

You can get the reset reason via 

uint32 CySysGetResetReason(uint32 reason);

Mr-HaleYa, 2022-07-31:
You shouldn't need to include CyLib.h as it is included by almost every other file. But if it throws an error during compile like "function not defined" then add `#include "CyLib.h"`.

---

function found in cores/asr650x/projects/PSoC4/CyLib.c [L2654](https://github.com/HelTecAutomation/CubeCell-Arduino/blob/68b01ab7bf04ab2fb4d41a01287dcf4b42defd97/cores/asr650x/projects/PSoC4/CyLib.c#L2654)
```c
/*******************************************************************************
* Function Name: CySoftwareReset
****************************************************************************//**
*
*  Forces a software reset of the device.
*
*******************************************************************************/
void CySoftwareReset(void)
{
    /***************************************************************************
    * Setting the system reset request bit. The vector key value must be written
    * to the register, otherwise the register write is unpredictable.
    ***************************************************************************/
    CY_SYS_AIRCR_REG = (CY_SYS_AIRCR_REG & (uint32)(~CY_SYS_AIRCR_VECTKEY_MASK)) |
                            CY_SYS_AIRCR_VECTKEY | CY_SYS_AIRCR_SYSRESETREQ;
}
```

function found in cores/asr650x/projects/PSoC4/CyLib.c [L2312](https://github.com/HelTecAutomation/CubeCell-Arduino/blob/68b01ab7bf04ab2fb4d41a01287dcf4b42defd97/cores/asr650x/projects/PSoC4/CyLib.c#L2312)
```c
/*******************************************************************************
* Function Name: CySysGetResetReason
****************************************************************************//**
*
*  Reports the cause for the latest reset(s) that occurred in the system. All
*  the bits in the RES_CAUSE register assert when the corresponding reset cause
*  occurs and must be cleared by the firmware. These bits are cleared by the
*  hardware only during XRES, POR, or a detected brown-out.
*
*  \param reason: bits in the RES_CAUSE register to clear.
*   CY_SYS_RESET_WDT       - WDT caused a reset
*   CY_SYS_RESET_PROTFAULT - Occured protection violation that requires reset
*   CY_SYS_RESET_SW        - Cortex-M0 requested a system reset.
*
* \return
*  Status. Same enumerated bit values as used for the reason parameter.
*
*******************************************************************************/
uint32 CySysGetResetReason(uint32 reason)
{
    uint32 returnStatus;

    reason &= (CY_SYS_RESET_WDT | CY_SYS_RESET_PROTFAULT | CY_SYS_RESET_SW);
    returnStatus = CY_SYS_RES_CAUSE_REG &
                  (CY_SYS_RESET_WDT | CY_SYS_RESET_PROTFAULT | CY_SYS_RESET_SW);
    CY_SYS_RES_CAUSE_REG = reason;

    return (returnStatus);
}
```

call `CySoftwareReset();` to reboot the device
call `Serial.println(CySysGetResetReason(0));` to view the reason for _Last_ reboot, this will return a number 0, 8, or 16

The three states CySysGetResetReason can detect are 

-   CY_SYS_RESET_WDT - WDT caused a reset, the returned value will be 0
-   CY_SYS_RESET_PROTFAULT - Occured protection violation that requires a reset, the returned value will be 8
-   CY_SYS_RESET_SW - Cortex-M0 requested a system reset, the returned value will be 16

closed 2022-07-31 by Mr-HaleYa