HelTecAutomation/CubeCell-Arduino #247: Example "LoRaWan_OnBoardGPS_Air530.ino" stops working after some time

505222621, 2022-08-01:
Board: CubeCell HTCC-AB02S GPS
Example: LoRaWan_OnBoardGPS_Air530
After running for a few minutes in the open field, the code gets stuck in the GPS segment

Mr-HaleYa, 2022-08-01:
are you using battery power?

505222621, 2022-08-02:
> are you using battery power?

no,using USB power

rodri16, 2022-08-22:
Same problem here...

505222621, 2022-08-25:
> This is because it constantly starts and stops the GPS and there is a memory leak bug in the GPS.end() function that is fixed with this commit: [85c0c75](https://github.com/HelTecAutomation/CubeCell-Arduino/commit/85c0c7542b4496658624e578f45d187c321bc8ad)

yes,GPS.begin -> serial-begin : malloc(buf),no free(buf),change to global variable or serial-end :free(buf)

closed 2022-08-25 by 505222621