HelTecAutomation/CubeCell-Arduino #252: documentation link down

Tomi05-k, 2022-09-21:
Hi, I want to integrate the cube cell into arduino. Unfortunately the link is dead. I need the json link for arduino boardmanager.

Greetings Tom!

ScholliYT, 2022-09-21:
The main docs [link](https://docs.heltec.org/en/latest/cubecell/index.html) seems to be down, however the docs are still up [here](https://docs.heltec.org/en/node/cubecell/quick_start.html#install-cubecell-relevant-framework).

The boardsmanager json file is [here](https://github.com/HelTecAutomation/CubeCell-Arduino/blob/master/package/package_CubeCell_index.json) on GitHub and you can use this URL: https://github.com/HelTecAutomation/CubeCell-Arduino/releases/download/V1.4.0/package_CubeCell_index.json

Andy1968T, 2022-10-22:
The json provided fails to resolve the issue of dead link for arduino board manager. Even with this seeming to be the correct json .
Links within the English github are most;y broken. Have i just purchased a board that is gaining support or losing support?
This link below fails to add the Heltec pill link.
 
https://github.com/HelTecAutomation/CubeCell-Arduino/releases/download/V1.4.0/package_CubeCell_index.json

ScholliYT, 2022-10-22:
Mh yeah, I guess parts of the links are broken (see #254). Until a fix arrives I can recommend using PlatformIO to program your Heltec CubeCell Boards: https://docs.platformio.org/en/stable/boards/asrmicro650x/cubecell_board_plus.html