HelTecAutomation/CubeCell-Arduino #261: floating point usage in radio.c

kritzikratzi, 2022-12-04:
radio.c uses floating point numbers (doubles, to be precise) to calculate the air time (maybe there are other places too), however it would be quite trivial to rewrite this computation to fixed point maths. 

the goal would be to get rid of the generated code for float/double support: 

 8035: 0000f7bd 1464 FUNC GLOBAL HIDDEN 2 __aeabi_ddiv
 8141: 0000f0d1 1772 FUNC GLOBAL HIDDEN 2 __aeabi_dadd
 8045: 0000ffb1 1252 FUNC GLOBAL HIDDEN 2 __aeabi_dmul
 8097: 0000ecc9 840 FUNC GLOBAL HIDDEN 2 __aeabi_fsub
 8279: 0000ea99 560 FUNC GLOBAL HIDDEN 2 __aeabi_fdiv

in total this accounts for 5.8kB of flash, which is imho quite a lot.

gasagna, 2023-02-11:
radio.c is taken from [LoraMac-node](https://github.com/Lora-net/LoRaMac-node). Can you open an issue there? there's more chance it will be addressed, and perhaps we'll have the fix here too at some point.

kritzikratzi, 2023-02-13:
the issue was fixed upstream in 2020.

gasagna, 2023-02-13:
Oh, I see. So yes, perhaps we could use an update to radio.c.