HelTecAutomation/CubeCell-Arduino #263: device does not wake up from sleep mode

jjanderson, 2022-12-21:
I am using this sketch - https://github.com/HelTecAutomation/CubeCell-Arduino/blob/master/libraries/LoRa/examples/LoRaWAN/LoRaWAN_Sensors/LoRaWan_OnBoardGPS_Air530Z/LoRaWan_OnBoardGPS_Air530Z.ino for my HTCC-AB02S.

I am powering the device with a battery, however the device does not seem to wake up. I have added the PWM1.h and PWM2.h libraries as well as the void pwmEnable()  as per below which is my full sketch. (all keys zero'd out - connecting is not my issue)

The device seems to be working fine when powered by USB. I have been reading and searching for solutions for 2 days now and despite there being many comments about this, I am unable to find a definitive answer or good example (or updated example sketch) to get this resolved, and as such am raising an issue here in desperation to try and get this issue resolved.

```
/* The example is for CubeCell_GPS,
 * GPS works only before lorawan uplink, the board current is about 45uA when in lowpower mode.
 */
#include "LoRaWan_APP.h"
#include "Arduino.h"
#include "GPS_Air530.h"
#include "GPS_Air530Z.h"
#include "HT_SSD1306Wire.h"
#include "PWM1.h"
#include "PWM2.h"

Air530ZClass GPS;
extern SSD1306Wire display;

//when gps waked, if in GPS_UPDATE_TIMEOUT, gps not fixed then into low power mode
#define GPS_UPDATE_TIMEOUT 120000

//once fixed, GPS_CONTINUE_TIME later into low power mode
#define GPS_CONTINUE_TIME 10000
/*
   set LoraWan_RGB to Active,the RGB active in loraWan
   RGB red means sending;
   RGB purple means joined done;
   RGB blue means RxWindow1;
   RGB yellow means RxWindow2;
   RGB green means received done;
*/

/* OTAA para*/
uint8_t devEui[] = { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };
uint8_t appEui[] = { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };
uint8_t appKey[] = { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };

/* ABP para*/
uint8_t nwkSKey[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };
uint8_t appSKey[] = { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };
uint32_t devAddr = (uint32_t)000000000000;

/*LoraWan channelsmask, default channels 0-7*/
uint16_t userChannelsMask[6] = { 0x00FF, 0x0000, 0x0000, 0x0000, 0x0000, 0x0000 };

/*LoraWan region, select in arduino IDE tools*/
LoRaMacRegion_t loraWanRegion = ACTIVE_REGION;

/*LoraWan Class, Class A and Class C are supported*/
DeviceClass_t loraWanClass = LORAWAN_CLASS;

/*the application data transmission duty cycle.  value in [ms].*/
uint32_t appTxDutyCycle = 300000;

/*OTAA or ABP*/
bool overTheAirActivation = LORAWAN_NETMODE;

/*ADR enable*/
bool loraWanAdr = LORAWAN_ADR;

/* set LORAWAN_Net_Reserve ON, the node could save the network info to flash, when node reset not need to join again */
bool keepNet = LORAWAN_NET_RESERVE;

/* Indicates if the node is sending confirmed or unconfirmed messages */
bool isTxConfirmed = LORAWAN_UPLINKMODE;

/* Application port */
[... 293 lines cut ...]
      }
    case DEVICE_STATE_SLEEP:
      {
        LoRaWAN.displayAck();
        LoRaWAN.sleep();
        break;
      }

    default:
      {
        deviceState = DEVICE_STATE_INIT;
        break;
      }
  }
}
void pwmEnable() {
  uint8 enableInterrupts;
  enableInterrupts = CyEnterCriticalSection();
  PWM2_BLOCK_CONTROL_REG |= PWM2_MASK;
  CyExitCriticalSection(enableInterrupts);
  enableInterrupts = CyEnterCriticalSection();
  PWM2_COMMAND_REG = ((uint32)(PWM2_MASK << PWM2_CMD_START));
  CyExitCriticalSection(enableInterrupts);
}
```

Koloblix, 2023-01-23:
Hi,

you have this here in your Code:

`/*the application data transmission duty cycle.  value in [ms].*/
uint32_t appTxDutyCycle = 300000;`

maybe you shud set this to a lower value, You send it for 300000ms into sleeop until next run

jjanderson, 2023-01-24:
300000ms is every 5 minutes. This should not be the issue here. Many LoRaWAN devices only communicate once a day. I have another setup where I use the same boards and transmit only every 30 minutes (1,800,000 Milliseconds) and this seems to work no problem. The issue is when I take it down from 5 minutes and lower... sampling every 1 minute for example is very very unreliable.