HelTecAutomation/CubeCell-Arduino #271: LoRaWan_interrupt example missing ~50% of transmits

metro69, 2023-04-07:
I am running the LoRaWan_interrupt example on a HTCC-AB01 v2 board.

Without any changes, I am losing ~50% of all USER_KEY presses: the interrupt happens, but the LoRaWAN transmit never happens.

In the serial output below, every � coincides with a missed transmit.

It looks like some other interrupt is interfering with the `LoRaWAN.send` command.

```
20:51:34.145 -> Sending data packet
20:51:34.145 -> unconfirmed uplink sending ...
20:51:37.715 -> 131106accel woke
20:51:37.715 -> Sending data�134036accel woke
20:51:40.631 -> Sending data�136381accel woke
20:51:42.991 -> Sending data packet
20:51:42.991 -> unconfirmed uplink sending ...
```

ScholliYT, 2023-04-08:
Are you being rate limited by your regional LoRaWAN parameters?

metro69, 2023-04-08:
I don't think so. It seems more a matter of the lorawan code generating interrupts that interfere with the serial.print and dropping the send request without any failure indication.

If I leave ~10s between button presses, transmits seem to work.

I would expect the API to be robust to signal the send call to not succeed if the system is not ready to accept another send.

ScholliYT, 2023-04-08:
Yep, the codebase isn't that good. And it is not well maintained at the moment. Guys from Heltec are not really responsive on this repository.