HelTecAutomation/CubeCell-Arduino #272: Example LoRaWan_OnBoardGPS_Air530.ino stops working with HTTC-AB02S  after hours

lyncker, 2023-04-08:
Hi, 
im trying to run the example (LoRaWan_OnBoardGPS_Air530.ino) from this repo with my GPS CubCell Board HTTC-AB02S.  After some hours (usually more then 12) the board stops working, which means it is not sending anything more to the TTN. I also can see that the serial out stalls ... 

I found some issues here saying there was some problem with the hardware serial buffer which is not beeing freed correctly , so I veryfied to install the version from this repository using the git command like described in the documenation... 
So now I should have the latest sources installed , but the problem persists. ... does anyone have an idea how to handle this ?

lyncker, 2023-04-12:
ok I ran this memcheck function tonight and here are the results after stop working some hours ago:
If I start the sketch calling the memceck it prints this: 

```
Heap Start:  0x200016c8, Biggest chunk: 1584
Stack Start: 0x20003fb7, Stack: 0x20003f87, Size: 48
```

the last printout bevor stop working was: 

```
Heap Start:  0x200016c8, Biggest chunk: 2132
Stack Start: 0x20003fb7, Stack: 0x20003f87, Size: 48 
```
I would expect that the biggest chunk gets smaller but it is growing...

jjanderson, 2024-02-26:
has anyone got a fix for this / or updated sketch please?