HelTecAutomation/CubeCell-Arduino #274: Radio.Init() Changes LED Pixel Behavior

Teku16, 2023-05-08:
Hello,

I am using a modified LoRa send code on an AB02S module, and when the following lines are used (required for our code), I lose all ability to control the pixel beyond the color green.

```
Radio.Init( &RadioEvents );
Radio.SetChannel( RF_FREQUENCY );
Radio.SetTxConfig( MODEM_LORA, TX_OUTPUT_POWER, 0, LORA_BANDWIDTH,
                                  LORA_SPREADING_FACTOR, LORA_CODINGRATE,
                                  LORA_PREAMBLE_LENGTH, LORA_FIX_LENGTH_PAYLOAD_ON,
                                  true, 0, 0, LORA_IQ_INVERSION_ON, 3000 );
Radio.SetSyncWord(0x13);
```

PlatformIO file:
```
[env:cubecell_gps]
platform = heltec-cubecell
board = cubecell_gps
framework = arduino
upload_port = COM5
upload_speed = 921600
monitor_port = COM5
monitor_speed = 115200
```

Libraries in use:
```
#include "LoRaWan_APP.h"
#include "Arduino.h"
#include "GPS_Air530Z.h"
#include "HT_SSD1306Wire.h"
#include "CubeCell_NeoPixel.h"
```

I don't think my code is relevant to the issue. I thoroughly tested via removing sections of my code, and the part above is what causes the LED to go only green, regardless of what I do (pixels.clear(), pixels.setPixelColor).

Teku16, 2023-05-08:
I also tried using the built-in `turnOnRGB(COLOR_SEND,0); //change rgb color RED` and I got a lime color. Violet gets white, yellow gets green, blue gets aqua.. They are all green based. When I remove the Radio references, I have no issues getting full spectrum RGB from either method. I have tried this on multiple AB02S units.

identeddie, 2024-02-23:
Did you find anything?

srengstorf, 2024-10-04:
I have 20 version 2.2 boards. On 10 boards the Neopixel LED looks a little different than on the other boards. On these boards there is a problem with the green LED. I think the problem is related to the fact that there are different batches of version 2.2. Is there a solution for this?