HelTecAutomation/CubeCell-Arduino #276: can't access https://heltec.cn/, so Arduino can't be used

rameyjm7, 2023-06-28:
https://heltec.cn/ doesn't work for me, is the site down? The arduino package manager JSON file also links there and get.exe, so they all fail to work.

Can you post the binaries somewhere accessible? At least on a google drive or something if you can't have them in the repo itself.

ristomatti, 2023-08-03:
The url seems to redirect to https://heltec.org/ which opens up fine in the browser at least.

rameyjm7, 2023-08-03:
Thanks, it looks like my laptop has the site blocked for some reason. I
accessed it on my phone and was able to get to that site. I'll assume this
is a problem I could work around with a different PC downloading the code
or something. Thanks.

  Jake