HelTecAutomation/CubeCell-Arduino #280: v1.6.0 release is missing package_CubeCell_index.json file

sysfu, 2023-09-04:
There's no package_CubeCell_index.json file listed in the [Assets for the 1.6.0 release](https://github.com/HelTecAutomation/CubeCell-Arduino/releases/tag/V1.6.0).

schroepfer, 2023-09-23:
Heltec! 
Please add this json or remove 1.6.0 
This makes so no sense.

amirna2, 2024-02-29:
This is becoming ridiculous. Also now it seems the download of 1.5.0 no longer works from Github CI. I have been continuously getting EOF errors since yesterday. When it has been working well before.

I can't use 1.6.0 because the package JSON file is missing

banthaherder, 2025-04-09:
In case anyone is still having trouble installing version 1.6.0, it's not officially published under the release's assets but is still bundled under the tag's source code. You can add it in the Arduino Board Manager using this URL:
https://raw.githubusercontent.com/HelTecAutomation/CubeCell-Arduino/refs/tags/V1.6.0/package/package_CubeCell_index.json