HelTecAutomation/CubeCell-Arduino #281: Provide source code for gcc toolchain

onitake, 2023-09-10:
This is a follow-up to #105 (and possibly #80 ).

As per [the conditions of the GPL](https://gcc.gnu.org/onlinedocs/gcc-8.5.0/gcc/Copying.html), you're obliged to provide source code for a software that contains GPL code, if requested. It doesn't matter if you modified the original code or not, but if you do, the modifications must be released under the GPL as well.

As the GNU Compiler Collection (gcc) is released under the the GPL, I'm formally requesting the source code for the toolchain downloadable from https://resource.heltec.cn/download/gcc-arm-none-eabi-8-2019-q3-update-linux.tar.gz .

Judging by the file names, it looks like you're distributing the GNU Arm Embedded Toolchain from https://developer.arm.com/downloads/-/gnu-rm#panel7a . If this is the case, I think it would be sufficient to copy the source package from ARM to your download server. This will ensure it remains downloadable when ARM removes the download or the ARM website is inaccessible for some reason.
Nonetheless, I think it would be polite that you mention in your documentation that the CubeCell SDK uses the GNU Arm Embedded Toolchain and link to the ARM website. Maybe also mention that new versions can be downloaded from there, if you want to support this.

Furthermore, I was wondering if you could release the source code for CubeCellflash, CubeCellelftool and flash6601.
As you mentioned in https://github.com/HelTecAutomation/CubeCell-Arduino/issues/80#issuecomment-618162467 , these tools contain trade secrets (a custom bootloader?) and you're not ready to release them as open source yet.
Has this changed?

onitake, 2023-09-10:
FYI: The PlatformIO documentation explicitly mentions the ARM GNU toolchain: https://docs.platformio.org/en/latest/platforms/heltec-cubecell.html#packages

echonet-capstone-nest, 2024-11-09:
Man, I would love to have this too. Or at least an aarch64 complied version of the rest of the toolchain. I'm wanting to run this on a SBC and mine is linux-arm. Can't use this toolchain on this SBC because of this.

onitake, 2024-11-09:
@echonet-capstone-nest Assuming you're doing this on a Linux system, you could try running the proprietary tools through Qemu emulation. You won't even need to set up a VM for this, just make sure Qemu userspace emulation is installed (`apt install binfmt-support qemu-user-static`) and run Cubecellflash et al in an amd64 container (e.g. `podman run -ti --rm --arch amd64 debian:stable`). More info: https://wiki.debian.org/QemuUserEmulation