HelTecAutomation/CubeCell-Arduino #287: CubeCell micros() issue in Arduino code

michapr, 2024-03-25:
We are expecting different problems with the CubeCell using the micros() function.

To reproduce:

```
#include <Arduino.h>

uint32_t last = 0;

void setup() {
    Serial.begin(115200);
    delay(2000);
    while(!Serial);
}

void loop() {
    uint32_t now = micros();
    Serial.println(now);
    if(now < last) {
        Serial.println("Stopped!");
        Serial.println(now);
        Serial.println(last);
        delay(900);
    }
    last = now;
}
```

`now `should never be smaller than `last`, yet we get the following output about every other x seconds :

```
12:38:27.451 > Stopped!
12:38:27.456 > 92596866
12:38:27.458 > 92596999
```

Interestingly, the last value that is printed, always ends in 999 or sometimes 998 or 997. 
However, removing the line Serial.println(now); makes it such that the problem doesn't happen... why?????

With `millis()` it runs fine - so something is wrong. Can you check it?

jgromes, 2024-03-31:
+1 since this actually showed up in two separate issues in RadioLib, [first here](https://github.com/jgromes/RadioLib/issues/1013) and then [second here](https://github.com/jgromes/RadioLib/issues/1041).

ellisgl, 2024-03-31:
Saw a note in the code about micros having ~1.7% error.  http://community.heltec.cn/t/cubecell-micros-function/985